jmcerrejon/PiKISS #193: Redream does not start

Rak1ta, 2023-10-13:
A startup error occurs when using your version of Redream. If you use [version](https://redream.io/download) v1.5.0-1080-g1053967, the launch is correct.
But I haven’t tested it on Bullseye armhf/aarch64, my ssd images are 240GB, it takes a very long time to rewrite copies.

https://redream.io/download/redream.universal-raspberry-linux-v1.5.0-1080-g1053967.tar.gz

```
OS: Debian GNU/Linux 12 (bookworm)
OS architecture: 64-bit
Kernel: aarch64 6.1.0-rpi4-rpi-v8
Device model: Raspberry Pi 4 Model B Rev 1.2 
Cpu name: Cortex-A72
Ram size: 3.61 GB
Raspberry Pi OS image version: 2023-10-10
Language: en_US.UTF-8
```

```
Press [ENTER] to run...
[INFO]  SDL     redream v1.5.0-1043-gd3a6df2cd appdir=/home/pi/games/redream
[INFO]          read /home/pi/games/redream/redream.cfg
[WARN]          failed to open /home/pi/games/redream/redream.cfg
libGL error: did not find extension DRI_Mesa version 1
libGL error: failed to load driver: vc4
libGL error: did not find extension DRI_Mesa version 1
libGL error: failed to load driver: vc4
libGL error: MESA-LOADER: failed to open swrast: ./mesa/aarch64-linux-gnu/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths ./mesa/aarch64-linux-gnu/dri, suffix _dri)
libGL error: failed to load driver: swrast
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  112
  Current serial number in output stream:  113
```

jmcerrejon, 2023-10-13:
Just checked on Debian Bookworm 32-bit installed from scratch and It works. I added a screenshot of the output (sorry for the quality).

[image]

I'll check it out on 64-bit soon.

Rak1ta, 2023-10-13:
@jmcerrejon I don't understand how it works, but I'm currently using Bookworm with X11 (i'm not comfortable using wayland right now). Could this use case affect the launch?
If update 1.5.0 works for you, you can update.

jmcerrejon, 2024-02-01:
I improved the script at 8761eebb1703808ebbbdfddc31541d64c44f628a.
Checked on 64-bit.

closed 2024-02-01 by jmcerrejon