jmcerrejon/PiKISS #196: Compiling Vulkan on Raspberry Pi 4

dolfelt, 2023-10-25:
I am trying to compile using piKiss and I keep running into the following Gallium error. I'm on bullseye 32-bit.

```../src/gallium/auxiliary/gallivm/lp_bld_init.c: In function ‘create_pass_manager’:
../src/gallium/auxiliary/gallivm/lp_bld_init.c:162:7: error: implicit declaration of function ‘LLVMAddScalarReplAggregatesPass’ [-Werror=implicit-function-declaration]
       LLVMAddScalarReplAggregatesPass(gallivm->passmgr);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/gallium/auxiliary/gallivm/lp_bld_init.c:163:7: error: implicit declaration of function ‘LLVMAddEarlyCSEPass’; did you mean ‘LLVMAddAnalysisPasses’? [-Werror=implicit-function-declaration]
       LLVMAddEarlyCSEPass(gallivm->passmgr);
       ^~~~~~~~~~~~~~~~~~~
       LLVMAddAnalysisPasses
../src/gallium/auxiliary/gallivm/lp_bld_init.c:164:7: error: implicit declaration of function ‘LLVMAddCFGSimplificationPass’ [-Werror=implicit-function-declaration]
       LLVMAddCFGSimplificationPass(gallivm->passmgr);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/gallium/auxiliary/gallivm/lp_bld_init.c:174:7: error: implicit declaration of function ‘LLVMAddReassociatePass’; did you mean ‘LLVMAddAnalysisPasses’? [-Werror=implicit-function-declaration]
       LLVMAddReassociatePass(gallivm->passmgr);
       ^~~~~~~~~~~~~~~~~~~~~~
       LLVMAddAnalysisPasses
../src/gallium/auxiliary/gallivm/lp_bld_init.c:175:7: error: implicit declaration of function ‘LLVMAddPromoteMemoryToRegisterPass’ [-Werror=implicit-function-declaration]
       LLVMAddPromoteMemoryToRegisterPass(gallivm->passmgr);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/gallium/auxiliary/gallivm/lp_bld_init.c:177:7: error: implicit declaration of function ‘LLVMAddConstantPropagationPass’; did you mean ‘LLVMAddDestination’? [-Werror=implicit-function-declaration]
       LLVMAddConstantPropagationPass(gallivm->passmgr);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       LLVMAddDestination
../src/gallium/auxiliary/gallivm/lp_bld_init.c:181:7: error: implicit declaration of function ‘LLVMAddInstructionCombiningPass’; did you mean ‘LLVMGetInstructionCallConv’? [-Werror=implicit-function-declaration]
       LLVMAddInstructionCombiningPass(gallivm->passmgr);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       LLVMGetInstructionCallConv
../src/gallium/auxiliary/gallivm/lp_bld_init.c:182:7: error: implicit declaration of function ‘LLVMAddGVNPass’; did you mean ‘LLVMAddCase’? [-Werror=implicit-function-declaration]
       LLVMAddGVNPass(gallivm->passmgr);
       ^~~~~~~~~~~~~~
       LLVMAddCase
cc1: some warnings being treated as errors
[8/301] Compiling C object src/gallium/auxiliary/libgallium.a.p/gallivm_lp_bld_tgsi_action.c.o
ninja: build stopped: subcommand failed.```

Any help appreciated!

actraiser30, 2023-12-21:
Vulcan compilation fails on pikiss probably some libraries/depends missing..

jmcerrejon, 2023-12-22:
Fixed at c159710ad784bd2d2dbef0e139f13bb834378579.
Tested on 32-bit RPi 4 & 64-bit RPi 5 with RPi OS from scratch.
Please check it out. I let the issue open

closed 2024-01-13 by jmcerrejon