jmcerrejon/PiKISS #197: Pi detection broken on some kernels

theofficialgman, 2023-11-12:
It appears that the custom /proc/cpuinfo text is not available in some kernels.
We (at Pi-Apps) have been receiving error logs from your software failing to install due to checking /proc/cpuinfo for information that does not exist
```bash
    local IS_RASPBERRYPI
    IS_RASPBERRYPI=$(grep </proc/cpuinfo 'BCM2708\|BCM2709\|BCM2835\|BCM2711')
    cd "$INSTALL_DIR" || exit 1

    if [[ -z $IS_RASPBERRYPI ]]; then
        echo "Sorry. PiKISS is only available for Raspberry Pi boards."
        exit 1
    fi
```

the users are running kernels that identify as `6.1.61-v8+` and `6.1.61-v8_16k+`. It is possible these are testing kernels and will be pushed to all users soon on piOS. Your software will also fail to install on pi5 regardless when following install instructions -> `curl -sSL https://git.io/JfAPE | bash`

Your method of checking is unstable and you should consider instead checking for the devicetree compatible string for matches

```bash
    if grep -q </proc/device-tree/compatible 'bcm2712\|bcm2711\|bcm2837\|bcm2836\|bcm2835'; then
        echo "Sorry. PiKISS is only available for Raspberry Pi boards."
        exit 1
    fi
 ```

jmcerrejon, 2023-11-12:
Thank.

It's weird. I installed it yesterday with no issues, but now I'm getting the message you mentioned. I've added the code you sent me but is not working. I removed that IF and I'm gonna check what's going on.

jmcerrejon, 2023-11-12:
Fixed. My fault 😅

Thanks for all.

closed 2023-11-12 by jmcerrejon