jmcerrejon/PiKISS #198: Diablo II+Lord of Destruction Expansion Set - No Sound - Pi 4B+

striderhiryux99, 2023-11-17:
Hello everyone. I'm currently using a Raspberry Pi 4B+ with 4gb of RAM (Linux 5.10.17-v8+ aarch64 Raspbian). I was able to obtain the required files to load Diablo II+The Lord of Destruction Expansion Set. Wine opened the window at 800x600 resolution, but when the Blizzard introduction screen began to load, there was no sound produced. I continued past the title screen where you select your warrior type and still didn't notice any changes. I don't want to try re-installing because I got this far would not want to lose the .mpq files. Did I miss any required steps? Please let me know. Thanks.

jmcerrejon, 2023-11-18:
Hi. 
Sorry, Diablo 2 is not supported on 64 bits. The point is how did you run an armhf executable on aarch64 OS 🧐
I could recommend you go to the Diablo 2 source code repository, but doesn't exist.
[Notaz](https://github.com/notaz) is the person I got the file. I'm pretty sure he compiled the binary and maybe can help you. PiKISS only installs apps.

Good luck!

closed 2023-11-18 by jmcerrejon

striderhiryux99, 2023-11-19:
You probably won't believe this, but I tried again on the Pixel Desktop,
and it loaded with music and voices! The only problem now, is that it shows
that gtk files are missing and it keeps showing messages about graphical
issues on the Wine message screen in the background. Is there a folder I
should browse through to get a runncomand.log so I can show you what is
going on? Please let me know. Thanks.