jmcerrejon/PiKISS #201: Pi-Apps installer broken

Botspot, 2023-12-13:
Pi-Apps runs this command to install piKiss: 
```bash
curl -sSL https://git.io/JfAPE | bash || error 'Failed to install piKiss!'
```

Your commit here, "[👌 IMPROVE: run piKiss.sh when install it](https://github.com/jmcerrejon/PiKISS/commit/33ede7565752b4c7406b19ba88d433b021c4a5ea)" broke the installer.
Now when installing PiKISS from Pi-Apps, users are greeted with the pikiss interface. This is bad because installs should be automatic and noninteractive.
Also, your internet connection check is failing on some systems. We have had two error reports sent in where pikiss installs fine but once it is run, your `check_internet_available` function incorrectly detects a lost internet connection.

This issue is requesting two changes:
1. To avoid running piKiss.sh at the end of installation, at least for pi-apps users.
2. To improve your check_internet_available function.
3. To verify that in the future, your changes do not break pi-apps again. This is the second time it was happened in the past few weeks (the other occasion was for issue #199)

Botspot, 2023-12-13:
Attached are the two logfiles we received.
[install-fail-piKiss (3).txt](https://github.com/jmcerrejon/PiKISS/files/13662996/install-fail-piKiss.3.txt)
[install-fail-piKiss (2).txt](https://github.com/jmcerrejon/PiKISS/files/13662997/install-fail-piKiss.2.txt)

jmcerrejon, 2023-12-13:
I don't know when was the moment in the time PiKISS depended on Pi-Apps. I don't want to be rude, but I had to add some code to make it compatible with **YOUR** software, but is not my business. You always can remove my software from Pi-Apps.

So, I'm going to fix for Pi-Apps this time, but If It's broken again in the future, I'm afraid you must come here again and notify me, and maybe I'll see what I can do.

Botspot, 2023-12-13:
> I don't know when was the moment in the time PiKISS depended on Pi-Apps. I don't want to be rude, but I had to add some code to make it compatible with **YOUR** software, but is not my business.

I would clarify that for both times something broke, it was due to a change you made that broke noninteractive installation. Ideally there would not need to be any specific pi-apps workarounds. For example, you would have encountered the same two problems if you tried packaging pikiss into a deb package, as in some cases those can be noninteractive.
Pi-Apps is not the issue, but Pi-Apps is affected by the underlying issue. I just wanted to clarify that.

> You always can remove my software from Pi-Apps.

I could, but before you start suggesting such things, please be reminded that Pi-Apps has brought in 27,393 users to your project. [source](https://github.com/Botspot/pi-apps-analytics/blob/5686bd9da740c426120bb9182ae07ea0b9879ea6/clicklist#L215)
I try to evaluate apps on the basis of usefulness without playing favorites. PiKISS is useful so it stays. Of course if it keeps breaking and users cannot install it, then fairness would force me to disable it or even remove it. Yes I have done this before to multiple other apps.
There is no reason it should have to come to that, though I am making the assumption that you value your userbase and donors, some of which only found out about PiKISS because it was on Pi-Apps.

jmcerrejon, 2023-12-13:
Fixed at d2ecd27155cc039b4892ac30883ad12b98e12710 & 617360b4a61e20325fc4ecd54f65e29119e718f6.

> I could, but before you start suggesting such things, please be reminded that Pi-Apps has brought in 27,393 users to your project.

I repeat I am not going to test that the changes I make will work in Pi-Apps. You are free to do whatever you want in consequence. Maybe I don't sound fair towards users who bounce my software through you, and I feel sorry for them, but I will not tolerate any threats.

closed 2023-12-13 by jmcerrejon