jmcerrejon/PiKISS #202: Retroarch fails to run on Bookworm

Silent-Hunter, 2023-12-28:
Trying to run Retroarch gives the error: 
`retroarch: error while loading shared libraries: libavcodec.so.58: cannot open shared object file: No such file or directory`

However, it is not installable:
```
 sudo apt install libavcodec58
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libavcodec58 : Depends: libaom0 (>= 1.0.0) but it is not installable
                Depends: libcodec2-0.9 (>= 0.9.2) but it is not installable
                Depends: libdav1d4 (>= 0.2.1) but it is not installable
                Depends: libvpx6 (>= 1.6.0) but it is not installable
                Depends: libwebp6 (>= 0.5.1) but it is not installable
                Depends: libx264-160 (>= 2:0.160.3011+gitcde9a93) but it is not installable
                Depends: libx265-192 (>= 3.2) but it is not installable
E: Unable to correct problems, you have held broken packages.
```
I understand if it's not compatible with Bookworm yet.

jmcerrejon, 2024-01-02:
@foxhound311 rescue me!

jmcerrejon, 2024-01-13:
It seems the issue is due RetroArch and lib deps from RPi OS 64-bit I can't handle. We need to wait for the guys to update the repository. I'll let the issue opened to remember this problem.

Silent-Hunter, 2024-01-13:
Bookworm seems to be using the newer libavcodec59 instead of libavcodec58. Wouldn't that mean building retroarch against libavcodec59?

jmcerrejon, 2024-01-13:
> Bookworm seems to be using the newer libavcodec59 instead of libavcodec58. Wouldn't that mean building retroarch against libavcodec59?

Maybe @foxhound311 can check it out

Rak1ta, 2025-01-12:
Is there anything I can do now to get the emulator to start?

Silent-Hunter, 2025-01-12:
> Is there anything I can do now to get the emulator to start?

None, that I can tell. I think we need it to be recompiled against libavcodec59.

Silent-Hunter, 2025-07-03:
Any progress on this? I've tried occasionally to build it myself, to no avail.

jmcerrejon, 2025-07-03:
> Any progress on this? I've tried occasionally to build it myself, to no avail.

No success right now. Trixie is approaching, so maybe with the new repository, we could compile it successfully.

foxhound311, 2025-07-03:
> Hopefully! Actually I should be a bit more clear, I got it to build, but I couldn't get most of the cores to work. They would segfault. I don't remember which ones worked and which didn't, but if you want me to I can try again and let you know what happens?

You gotta also compile cores individually. It’s a real time consumer.

Silent-Hunter, 2025-07-03:
Ah yeah, I couldn't get most of them to compile.

Jojo-A, 2025-11-16:
Hi,
any news on that? Or is there anything I can try/do on my own?

Jojo-A, 2025-11-17:
I just saw that RetroArch is available in the official sources for Raspberry OS Trixie.
It installs successfully, but without cores of course.

SNES core libretro-snes9x is also avauilable, unstalls successfully and runs very good.

So maybe it would be a preferable solution for PiKISS to install Retroarch and cores independant from each other?

jmcerrejon, 2025-11-30:
Closed with a2d96143306baa60cce7a30ec6a0c25a02273dde
I'm gonna use the Debian repository version for now. I'll try to compile new cores in the future.
Thank you all!

closed 2025-11-30 by jmcerrejon