jmcerrejon/PiKISS #203: Error with Pi5 after updating Mesa to 23.3.1

darksaviorx, 2023-12-31:
I'm using RetroPie installed on top of bookworm 64bit lite and after updating the Mesa drivers to 23.3.1, Emulationstation no longer loads. 
`libEGL fatal: DRI driver not from this Mesa build ('23.3.1 (git-6cdc2b044d)' vs '23.2.1-1~bpo12+rpt2')`

jmcerrejon, 2024-01-02:
Did you try another Mesa version? With the script, you can use another two options: legacy from repo or latest compilation.

Another possible issue is that libEGL should be compiled using the version you have.

darksaviorx, 2024-01-03:
I tried the latest version right now and it gave me a similar error message. 
I'm not sure how to compile libEGL with the new version of Mesa.

actraiser30, 2024-01-03:
RetroPie is experimental most of it's features are not working correctly

closed 2024-01-13 by jmcerrejon