jmcerrejon/PiKISS #204: aborting on install??? pi5 bookworm

NossieUK, 2024-01-01:
errrr any thoughts@?

`curl -sSL https://git.io/JfAPE | bash`

`Aborting...
pi@Europa:~ $   `

theofficialgman, 2024-01-02:
see https://github.com/jmcerrejon/PiKISS/issues/201

due to continual upstream changes, pi-apps has now implemented an installer natively so that the error you reported can never happen again https://github.com/Botspot/pi-apps/commit/e0aa01cc89459930830c9f06eb72c8d012262d0b

jmcerrejon, 2024-01-02:
### My setup

Board: Raspberry Pi 5 8Gb
OS: Raspberry Pi OS Booksworm updated 64-bit

It works on my side.

[image]

Send more info or try it again.

closed 2024-01-13 by jmcerrejon