jmcerrejon/PiKISS #206: Dolphin fails with Segmentation Fault on fresh install Raspberry PI 4 Model B

White-waluigi, 2024-01-14:
I have a RPI 4 Model B and installed PiKiss. I tried to install dolphin but it gave me a Segementation fault:

Press [ENTER] to run the emulator...
./scripts/emus/dolphin.sh: line 14:  3145 Segmentation fault      /usr/local/bin/dolphin-emu

I tried to find more Information but unfortunately a log file was not created.

It did create a file .dolphin-emu/Wii/sys/uid.sys which is empty. I did not find a log.

I also get Segementation Fault if I run it directdly

jmcerrejon, 2024-01-14:
I guess you are trying on RPi OS 32-Bit (not available on 64-bit).

White-waluigi, 2024-01-14:
Tried on both. 64 didn't even install, as you said

jmcerrejon, 2024-01-14:
I'm burning the OS 32-bit version and check it out. Keep you informed.

jmcerrejon, 2024-01-14:
I'm checking my script and is outdated (2020!). It was not compiled by myself, but I see instructions to do it and It does not seem difficult. I'll try to compile it next week. Try another emulator in the meantime 😅

White-waluigi, 2024-01-22:
Got it running with these instructions on 64bit:

sudo apt update

sudo apt upgrade

sudo apt install --no-install-recommends ca-certificates qtbase5-dev qtbase5-private-dev git cmake make gcc g++ pkg-config libavcodec-dev libavformat-dev libavutil-dev libswscale-dev libxi-dev libxrandr-dev libudev-dev libevdev-dev libsfml-dev libminiupnpc-dev libmbedtls-dev libcurl4-openssl-dev libhidapi-dev libsystemd-dev libbluetooth-dev libasound2-dev libpulse-dev libpugixml-dev libbz2-dev libzstd-dev liblzo2-dev libpng-dev libusb-1.0-0-dev gettext

sudo apt install mesa-vulkan-drivers

sudo apt install qt6-base-dev qt6-svg-dev qt6-base-private-dev

git clone https://github.com/dolphin-emu/dolphin.git dolphin-emu

cd dolphin-emu

git submodule update --init

cmake ..

make -j4

sudo make install

sudo reboot

jmcerrejon, 2024-01-25:
Fixed at 3f4d237de31aeaf8fd3c95e9c383ae00a0505a9b

Thank for you help, @White-waluigi!. ☺️

closed 2024-01-25 by jmcerrejon