jmcerrejon/PiKISS #207: "cannot currently show the desktop" after compiling Vulkan.

Gabiiofficial, 2024-01-28:
I'm running the latest Raspberry Pi OS 64 bit on a Raspberry Pi 5 8 Gb.

Yesterday i compiled the vulkan drivers using PiKISS. Everything worked fine, i turned off my pi, and went to sleep.
Today i turned on my pi, and my vnc said "cannot currently show the desktop". I tried to connect my pi to a monitor, and it received input, but it showed a black screen. I can ssh to the pi.

Gabiiofficial, 2024-01-28:
edit: i was able to fix it by going in ~/mesa_vulkan/build folder and running 'sudo ninja install', as mentioned in issue #165

jmcerrejon, 2024-01-28:
It's weird because the command should be run 🤷‍♂️ 
I'll check it out next week. Thanks for reporting it.

jmcerrejon, 2024-01-31:
Tried on my OS with the latest RPi OS 64-bit and It's OK when rebooted. Send you a screenshot

[image]

Next time I edit the script, I'll include a check for installed files. In the meantime, the script works.

Thanks for the report, anyway. 😊

closed 2024-01-31 by jmcerrejon

NossieUK, 2024-02-01:
I've done this recently too sudo ninja install works until the next restart then I have to do it again - any idea?

jmcerrejon, 2024-02-12:
I think the problem with the Vulkan driver is If you update your distro.

In my case, the desktop is shown always on each reboot, **but** If I run `apt update` and install the packages provided on it, then after reboot, the system hangs on the terminal trying to enable the desktop environment. No matter what were the packages installed.

What is the solution to this problem? I don't know yet. I'll dive into that.

NossieUK, 2024-02-12:
I followed your thoughts and tried the stable one and it didn't do it every reboot but as you say on an update \o/

jmcerrejon, 2024-02-22:
Maybe this could be a "partial" fix: https://misapuntesde.com/2024/02/run-a-script-after-apt-update-on-debian.html

What do you think about it?