jmcerrejon/PiKISS #209: Remove vulkan script

theofficialgman, 2024-03-20:
your vulkan script breaks users installs.

```
Vulkan driver not found. Exiting...
E: Problem executing scripts DPkg::Post-Invoke '/home/tonimous/mesa_vulkan/reinstall-vulkan-driver.sh'
E: Sub-process returned an error code
Apt finished.
Failed to install the packages!
APT reported these errors:
E: Problem executing scripts DPkg::Post-Invoke '/home/tonimous/mesa_vulkan/reinstall-vulkan-driver.sh'
E: Sub-process returned an error code
```

you have no buisness including such a script for users that know no better. install `mesa-vulkan-drivers` which is the supported and maintained version from the distro and be done with it. it gets updated (on piOS) when the developers feel it is best to do so.

theofficialgman, 2024-03-20:
Have you even read your own reinstall function?
You shouldn't ever be doing this but it is so obviously broken
```
#!/bin/bash

function reinstall_vulkan_driver() {
    readonly BUILD_MESA_VULKAN_DRIVER_DIR="/home/ulysess/mesa_vulkan/build"

    if [[ ! -d $BUILD_MESA_VULKAN_DRIVER_DIR ]]; then
        echo "Vulkan driver not found. Exiting..."
        exit 1
    fi

    cd $BUILD_MESA_VULKAN_DRIVER_DIR || exit 1
    echo "Reinstalling Vulkan driver..."
    sudo ninja install > /dev/null 2>&1
    echo "Vulkan driver reinstalled!."
}
reinstall_vulkan_driver
```
so all users are expected to be named `ulysess` now?

jmcerrejon, 2024-03-21:
You are right. I'll fix the reinstall script.

Talking about removing the script, I give the user three options to install the Vulkan drivers and warn about the consequences. Maybe I could change the recommendation to the option that install the driver from the repo.

I think Linux means you have choices to do what you want, even If you break your OS, so I'll keep the script, and the three options.

Of course, you can choose not to use the script, but let other users can choose if theyvwant to use the script or not.

jmcerrejon, 2024-03-21:
Fixed at 611162c52e444363a375974792c8e899334dcec0

Thank you!

closed 2024-03-21 by jmcerrejon

Botspot, 2024-03-28:
@jmcerrejon I'm not sure if you completely understood the situation from what @theofficialgman said, but yesterday we were flooded with error reports that all showed `apt` has been broken by your vulkan-reinstall script on a number of users' installs.
[install-fail-PPSSPP_PSP_emulator.txt](https://github.com/jmcerrejon/PiKISS/files/14783999/install-fail-PPSSPP_PSP_emulator.txt)
[install-fail-Reaper.txt](https://github.com/jmcerrejon/PiKISS/files/14784000/install-fail-Reaper.txt)
[install-fail-Reaper2.txt](https://github.com/jmcerrejon/PiKISS/files/14784001/install-fail-Reaper2.txt)
[install-fail-Steam.txt](https://github.com/jmcerrejon/PiKISS/files/14784002/install-fail-Steam.txt)
[install-fail-Steam2.txt](https://github.com/jmcerrejon/PiKISS/files/14784003/install-fail-Steam2.txt)
[install-fail-Steam3.txt](https://github.com/jmcerrejon/PiKISS/files/14784004/install-fail-Steam3.txt)
[install-fail-Steam4.txt](https://github.com/jmcerrejon/PiKISS/files/14784005/install-fail-Steam4.txt)

For these affected users, NOTHING done by apt will succeed from now on. Apt will fail during any operation due to this syntax error in your script.

As a maintainer of Pi-Apps, I have high standards for reliability. Everything we offer is well-tested ahead of time. For example, before we push an update to an app, we have an automated system that checks if the update works on PiOS Buster, Bullseye, and Bookworm, both 32-bit and 64-bit, as well as 5 flavors of Ubuntu.

Users come first. We serve them.

Occasionally, somebody blames Pi-Apps for breaking their system, so we try very hard to keep that from ever happening. What happened yesterday was unfortunate. Several people who were using your vulkan script one day found their systems not working anymore, and all they knew to do was to send us an error report. I am not aware of any error-reporting in PiKISS, so without our "send report" button, it is likely that even now, you would still not know about the bug in the first place.

We should be able to support each other. PiKISS solves a different problem than Pi-Apps, and together we can reach more people and work to bring Raspberry Pi and FOSS to the mainstream world.

Partnership is good. But it takes **cooperation** from both sides.

This screenshot does not show cooperation, it shows someone unwilling to help when asked.
[image]
And this is not the only example. Here are a few others.
[image]
[image]
We have always been willing to cooperate the whole time. In fact, after you made a change to the PiKISS installation script that broke headless usage for the third time, we just [rewrote the script](https://github.com/Botspot/pi-apps/commit/e0aa01cc89459930830c9f06eb72c8d012262d0b) that installs PiKISS, instead of asking you to fix it.

We have always been willing to help, and we have taken the time to report issues as they come up and clarify what needs fixing. And in return, you have been combative, falsely blaming Pi-Apps for causing the issues, and overall just being uncooperative.

A while back I said this on issue #201:
> I try to evaluate apps on the basis of usefulness without playing favorites. PiKISS is useful so it stays. Of course if it keeps breaking and users cannot install it, then fairness would force me to disable it or even remove it. Yes I have done this before to multiple other apps.

You seemed to think that was a threat, but that was not a threat. It was the honest, fair truth.

Enough has been said. PiKISS and PiKISS-GUI have both been [removed](https://github.com/Botspot/pi-apps/commit/6e904e0418d742fbdacd2864b24497b0c6584d8f) from Pi-Apps earlier today. All users with PiKISS installed will receive a notification on the next update that explains why we no longer offer the app, and asks permission to uninstall it.
Also, I will be linking to this statement in our Discord server for others to read if they are curious. I would advise you to not delete this post.

jmcerrejon, 2024-03-28:
What you can't do is come to my house and tell me with bad words my music is too loud. I tell you and @theofficialgman I need more info to fix the issue. He did not provide any other info and came again with the idea of removing the script code. All code developed by humans has bugs, and the solution is not to remove the application. If you don't like or have a bug... Hey! Guess what?! You can contribute! Send me a PR and I'll merge with a happy smile.

I changed the recommended option to use the default Vulkan driver in the script. If you are a brave person, try the alternative, but It's not for all users. Is that not good for your users? So then, I agree with you about the decision to remove PiKISS from PiApps.

Best wishes for your app, you are doing better work than me. I'll try to do my best with the Vulkan script next week.

marcalexander2001gerbert-droid, 2025-08-25:
https://images.bestfans.com/images/z2y11nNd2g/geQy4DVveD/1/0/5c31ada9da1980f75ff663b6dcfa12ae.webp?h=GvX45Fsp41s58IN-rEsMWA&e=1756951200&s=1600