jmcerrejon/PiKISS #210: Remove vulkan script (v2)

theofficialgman, 2024-03-27:
```
Unpacking pi-apps-9fd06f44 (1.0) ...

Reinstalling Vulkan driver...
Vulkan driver reinstalled!.
sh: 1: Syntax error: Unterminated quoted string
E: Problem executing scripts DPkg::Post-Invoke 'fi''
E: Sub-process returned an error code
Apt finished.
```

I am really sick of this... remove these scripts. You clearly cannot be trusted to not continue breaking users installs

jmcerrejon, 2024-03-27:
You have an issue with a bug caused by the old code. By the way, I don't give support to installations through PiApps. So you have two options: Be kind to developers who help you in some tasks (even when bugs appear) and give more information about your issue.

It's weird that according to your GitHub profile, you are not a newbie with Linux and I'm pretty sure you could fix the issue in your OS.

I repeat you, If you don't like my app, don't use it. That's all.

theofficialgman, 2024-03-27:
This isn't an installation of pi-kiss through pi-apps. This is a user error report from your buggy code on an entirely separate app.

theofficialgman, 2024-03-27:
> Be kind to developers who help you in some tasks

Your buggy code is not helpful. I have asked you to remove it.

jmcerrejon, 2024-03-27:
I tried. I’ll close the issue. You are not giving more information. Please, uninstall PiKISS from your OS.

closed 2024-03-27 by jmcerrejon