jmcerrejon/PiKISS #216: Unreal 99 install issues

Realinferno, 2024-04-13:
Good evening, 

I use piKiss on my Raspberry Pi 4B (4 GB), running Bullseye 32-bit OS. I installed a few applications successfully (e.g., Starcraft) but ran into some issues when installing UT99.

I followed the instructions in the YouTube video “PiKiss installs today… Unreal Tournament 99 on Raspberry Pi”. 

When I start the installation, everything goes fine until the very end, when it says “Generating icon…” Press [enter] to go back to the menu…

When I press enter, I get the following error: “Something is wrong, aborting…”

Pressing enter again, takes me back to the PiKiss menu. The folder /games/ut99 is successfully created, including all files and the icon in the start menu. After copying my own game files to the directory (according to the YouTube video), I get the following error when starting the game (clicking the icon): “Failed to execute child process /home/Meijering/games/ut99/SystemARM64/ut-bin” (no such file or directory). I checked the path and it does exist, including the file ut-bin. 

Any help is appreciated! In any case: thank you for this amazing piece of software!

[image]

jmcerrejon, 2024-04-13:
You mention your OS is 32-bit, and It's only supported for 64-bit architecture 😅

Realinferno, 2024-04-13:
Thanks for your very quick reply!

Ah, of course (systemARM64…). I somehow did not perceive this and just assumed piKiss as a whole was for 32-bit (BTW, the GitHub does state that bullseye 32-bit is the ideal stage). If I’m not mistaken, it is not explicitly stated in piKiss that UT99 is 64-bit only?

In any case, thanks for your assistance and all the work you put into piKiss!

jmcerrejon, 2024-04-13:
You are right. 99% of apps run on Bullseye 33-bit, but I’m migrating all to 64-bit.
Let me check If I can do anything to compile for 33-bit next week. I let this issue open until I can tell you a piece of good/bad news 😉

Realinferno, 2024-04-14:
Wow, that’s very generous of you! Naturally, I completely understand if it just won’t work for 32-bit or if it’s too much hassle to compile for 32-bit.

jmcerrejon, 2024-04-18:
Hi @Realinferno 

After some research on their repo [here](https://github.com/OldUnreal/UnrealTournamentPatches), I couldn't get a source code to compile. Maybe some other repository is somewhere to be compiled for this game.

I encourage you to look it up and give it to me if you really want to play it in armhf.

Good luck! 😄

closed 2024-04-18 by jmcerrejon

Realinferno, 2024-04-18:
Hi @jmcerrejon,

Thanks for doing the research! I wasn’t expecting a 32-bit version of the game. Nonetheless, if I happen to see a repository, I will pass it through.

Keep up the good work with PiKISS 👍