jmcerrejon/PiKISS #217: Enhancement:  Script -> UxPlay (successor to RPiPlay)

steveredden, 2024-04-24:
[https://github.com/FDH2/UxPlay](https://github.com/FDH2/UxPlay) is maintained, and operates on the latest bookworm-based releases, as well as 32-bit and 64-bit.

RPiPlay has not gotten any love lately.

jmcerrejon, 2024-04-25:
I just checked today in the morning.

My first impression: It's not very usable using RPi 5 on RPi Bookworm 64-bit + Wayland. After passing some arguments to the binary, I can see the screen, but It fails when I try to rotate the device. For a change, I think the guilty is the OS and/or Wayland because the older repository _RPiPlay_ ran flawlessly. Now you need to handle multiple arguments according to your settings (Rpi version, Video driver, Desktop environment...).

So, I don't know what to do. 😅 I prefer giving support to  _UxPlay_ because is updated, but I need to think about how to make it usable for multiple config/variations.

steveredden, 2024-04-25:
There's definitely some issues with both --> I haven't been able to get rpiplay to work at all... though i didn't dig into it too much.

I did spend more troubleshooting on uxplay, as it seemed newer.

I opted to build from the source as the raspbian repos were using an outdated version

There's also a gstreamer patch i had to do to make video flipping work:  https://github.com/FDH2/UxPlay/issues/287

And i created a service to start using the following: `/usr/local/bin/uxplay -a -r R -vs kmssink -avdec -fs -n Garage-UxPlay -nh` -- this worked the best on an RBPI 4B with a 64bit bookworm full install

jmcerrejon, 2024-04-27:
It works on my side with your guide (thank you!), but changing -vs with waylandsink (I'm using wayland). I need a way to know If the user is using Wayland or not.

jmcerrejon, 2024-06-09:
Closed at 68ab97e9873282d49f40fb92c8bba789fda462dc

Thank you!

closed 2024-06-09 by jmcerrejon