jmcerrejon/PiKISS #220: PCem for Raspberry Pi 4/5

actraiser30, 2024-06-10:
Hello 

Is it possible to add this??

https://github.com/sarah-walker-pcem/pcem

It's like dosbox but way better 

Thanks

jmcerrejon, 2024-06-10:
I can try to compile it! Keep you informed

jmcerrejon, 2024-06-10:
I'd like to test it more deeply, but after some caveat,... It works!

[image]

I hope to include on PiKISS soon. 🎉

actraiser30, 2024-06-10:
Thx a lot great news I will wait to test it

actraiser30, 2024-06-10:
Since you wanted to test it more,here is another one only when you have time it's similar 

https://github.com/86Box/86Box

Thanks again

Rak1ta, 2024-06-11:
@jmcerrejon It so happened that I myself tried PCem on raspberries for a long time. It works perfectly, even with "powerful" hardware, but in the OS (Win98) itself memory errors periodically occur during use. 
This is for those who want to try playing around with desktop OS.

jmcerrejon, 2024-06-11:
> 
> This is for those who want to try playing around with desktop OS. 

Maybe 86Box is a more updated project, with improvements to handle better memory leaks.

I'll add your info about Win98 as a warning to the script for all users, If you don't mind.

Rak1ta, 2024-06-11:
@jmcerrejon ok, but I'll try to install OS from scratch the other day

actraiser30, 2024-06-12:
May e it needs this

https://archive.org/details/pcem-v-17-roms

jmcerrejon, 2024-06-12:
> When I tried the windows version on my laptop with windows 98 i did not get a single error, Maybe it needs this specific files;
> 

v18 has new devices supported according to the [changelog](https://github.com/sarah-walker-pcem/pcem/blob/dev/CHANGELOG.md). Do you have a more recent link to those roms files?

Rak1ta, 2024-06-12:
@jmcerrejon This doesn't affect anything.  The list of ROMs with V17 is quite enough.

actraiser30, 2024-06-12:
> > https://archive.org/details/pcem-v-17-roms
> > https://github.com/BaRRaKudaRain/PCem-ROMs
> 
> v18 has new devices supported according to the [changelog](https://github.com/sarah-walker-pcem/pcem/blob/dev/CHANGELOG.md). Do you have a more recent link to those roms files?

V18 is not out yet,there aren't any rom files for v18
Maybe what you see on the changelog is just work in n progress,or a very early beta version.

V17 is the latest.

jmcerrejon, 2024-06-16:
Ready at 5be54bcc8cf526de8300251496dde69dce199b10! 

I pretend to improve the usability of this script by adding a .iso with basic software.

Maybe in the future, I could add custom PC configurations to deploy for example MS-DOS, Windows, or BeOS (yesss!!) in a snap!

Thanks @actraiser30 & @Rak1ta ❤️

actraiser30, 2024-06-16:
Thanks a lot,yes this auto deploy function to make windows 95/98 install super easier would be amazing,and also the basic software "iso"

Thanks again

Rak1ta, 2024-06-19:
@jmcerrejon your nickname came to me again 
[image]

actraiser30, 2024-06-19:
Maybe he is a fan like me with this

[image]

Rak1ta, 2024-06-19:
@jmcerrejon 

`~/.pcem` - Is it possible to leave customized user configs from deletion?

jmcerrejon, 2024-06-19:
> @jmcerrejon your nickname came to me again!

Mmm, I didn't know that. I'll add a regex to change the user's name in the script, because If I omit the file pcem.cfg, ROMs, and others are not found.

jmcerrejon, 2024-06-19:
> Maybe he is a fan like me with this 😉

Of course @actraiser30! My nickname was chosen because of that TV series. 😉

actraiser30, 2024-06-19:
Amazing series and the music is even better 
Still watching it sometimes.

Rak1ta, 2024-06-19:
> > @jmcerrejon your nickname came to me again!
> 
> Mmm, I didn't know that. I'll add a regex to change the user's name in the script, because If I omit the file pcem.cfg, ROMs, and others are not found.

Is it possible to do this here? I don't know if it was fixed or not.
https://github.com/jmcerrejon/PiKISS/issues/192

Rak1ta, 2024-06-19:
I promised to check how the program works and encountered a problem that the version collected from the official website on Debian 12 in my case crashes with an error when saving the new configuration. Although everything worked on Debian 10.
I don't know what the problem is, I used the pi-kiss option. Actually, a demonstration of work on Socket 7 and Slot 1. It's too much for raspberry, but I was too lazy to reinstall, I copied everything from a laptop.
Everything works, thank you!

[image]
[image]

jmcerrejon, 2024-06-19:
Nice to see Clippy again! 😂

actraiser30, 2024-06-19:
Rak1ta

How is the performance in raspberry pi with windows games?

Have you tried any age of empires 2 or StarCraft?

actraiser30, 2024-06-19:
> Nice to see Clippy again! 😂

Forgot to say I'm living close to Ithaca island in Greece 😁

jmcerrejon, 2024-06-19:
I studied Greek for a year a long time ago (don't ask me anything 😅 ).
Fixed at 3b2b6ef3b6feae6d545201ddf0a5888135662f97 Thank you!

closed 2024-06-19 by jmcerrejon

actraiser30, 2024-06-19:
> @actraiser30 I have not tried such. I only play games from consoles in RetroArch, some games through Wine/Box86 (but there is no point when there is a PC nearby). Wine works great. PCem is interesting to me how it all works and one program for drawing electrical diagrams for the institute 😅.

Is it possible to try with Pcem and within win98 Age of empires 2 demo ???

With dosbox pure and dosboxX in win98 was quite slow I'm curious how it's going with pcem 

https://archive.org/details/AgeofEmpiresIITheAgeofKings_1020

Thanks

Rak1ta, 2024-06-19:
@actraiser30 
I have Pi4. I can take 5, but I need a case from DeskPi, I'm uncomfortable without it.
If you follow these [system requirements](https://gamesystemrequirements.com/game/age-of-empires-ii-age-of-kings) (Pentium 166MHz, Super VGA), i can say that it works at 35% speed. The sound slows down because of this. You can probably build on older hardware, but there is already a complex BIOS for me and I need to re-understand the compatibility of devices.
In the usual Win98 unplayable, The Pi5 probably should be better. But my CPU load is only 50%. 
I hope I fully told the maximum that can be squeezed out of a raspberry.

[image]