jmcerrejon/PiKISS #226: Request: Any chance of Half-Life 2 being added?

CraigsBar, 2024-12-10:
This has been proven to work on Raspberry Pi 5 when running Ubuntu OS, Would it be possible to incorporate it here for easy installations on devices running Raspberry Pi OS.

jmcerrejon, 2024-12-10:
O tried with no success. Could you provide me any link with resources about it?

actraiser30, 2024-12-11:
Help on the way

https://www.jamesfmackenzie.com/howto/how-to-install-half-life-2-halflife-2-on-raspberry-pi/

CraigsBar, 2024-12-11:
I got it to compile on Raspberry Pi OS. But the graphics and character set are screwed up. 
[image]
[image]

actraiser30, 2024-12-11:
Try this in config.txt

Change the dtoverlay=vc4-kms-v3d 

To dtoverlay=vc4-fkms-v3d

And the opposite

CraigsBar, 2024-12-11:
Unfortunately that made no difference to the corrupted graphics/text.
[image]
[image]

actraiser30, 2024-12-11:
Try that,if you compile it in 64 bit OS  do it in 32bit OS or the opposite again.

jmcerrejon, 2024-12-11:
I already tried it (you can check [this](https://www.patreon.com/posts/108160432?utm_campaign=postshare_creator) open post at my Patreon. 

My issue was I only got 10-15 fps, and glitches in graphics as you commented.

For sure is some library used on RPi OS that is working on Ubuntu (Maybe the cause that guy compiled it there)

CraigsBar, 2024-12-11:
Interestingly I created an ubuntu install on an SD card this morning and compiled it there (my Pi's both normally boot from NVMe SSD's) and I get the same issues. Tonight I will try to get it to compile on my OpenSuSE laptop (Lenovo Thinkpad T460 so nothing special) and see how that goes. I wonder if it is a wider issue with the source on github and recent Linux environments.

CraigsBar, 2024-12-11:
> What about try it in wine from pi apps?

I find valve proton is more game compatible than native wine, and
unfortunately that crashes at startup.

actraiser30, 2024-12-11:
What about testing counterstrike source?

actraiser30, 2024-12-11:
Tested various games with wine on Pi4 even painkiller was playable if you lower the graphics
Nosferatu was very slow thought.

actraiser30, 2024-12-11:
What about this?

https://m.youtube.com/watch?v=xp-W1Axn8TM

jmcerrejon, 2024-12-11:
I think it’d be great to focus on running native apps on PiKISS without relying on Wine or Proton for games. That way, we can keep things simple and straightforward! 😉

actraiser30, 2024-12-12:
Native ports are better, yes,what about assault cube game can be build/work??

CraigsBar, 2024-12-13:
[image]
So it's the same on OpenSuSE on intel x64 (core i5)

Lohann, 2024-12-19:
Hi, I believe the Graphics issues is due an update in the Half-life from Steam, which is not supported by the current version of the source-engine:
https://github.com/nillerusr/source-engine/issues/397

Must test using a version prior to November 16 2024 release.

Lohann, 2024-12-20:
Is possible to download an older version from Steam from here:
https://steamdb.info/depot/224/manifests/

Using this tool:
https://github.com/SteamRE/DepotDownloader

Lohann, 2024-12-20:
Finally, I manage to get half-life 2 working natively on my **Rapsberry PI 5** with **Ubuntu 24.04**:
```bash
# Install dependencies
sudo apt-get install -y \
    unzip \
    wget \
    git-all \
    build-essential \
    pkg-config \
    ccache \
    libsdl2-dev \
    libfontconfig1-dev \
    libopenal-dev \
    libjpeg-dev \
    libpng-dev \
    libcurl4-gnutls-dev \
    libbz2-dev \
    libedit-dev

# Clone source-engine
cd ~/Downloads
git clone https://github.com/nillerusr/source-engine.git --recursive
cd source-engine

# Build Source-Engine, takes about ~22min to build on my RPi 5
python3 ./waf configure -T release --prefix=hl2 --build-games=hl2 --disable-warns
python3 ./waf build -p -v 
python3 ./waf install

# Download Half-Life 2, only the release from 17 November 2023 works:
# https://steamdb.info/patchnotes/12694556/
# Obs: Requires an steam account with the game.
cd ~/Downloads
wget https://github.com/SteamRE/DepotDownloader/releases/download/DepotDownloader_2.7.4/DepotDownloader-linux-arm64.zip
unzip DepotDownloader-linux-arm64.zip
./DepotDownloader -app 220 -depot 221 -manifest 3666218991449795038 -username <steam_username>

# Copy files to steam directory
rsync -ah --progress ./depots/221/16557524/hl2/* ~/Downloads/source-engine/hl2/hl2
cp ./depots/221/16557524/platform ~/Downloads/source-engine/hl2/

# start the game
cd ~/Downloads/source-engine/hl2
./hl2_launcher
```

jmcerrejon, 2024-12-20:
Thank you so much, @Lohann!

After encountering some issues with the commands I was using, your guide was incredibly helpful. I have successfully compiled the engine and obtained the correct game data files on Raspberry Pi OS 64-bit.

[image]

Now, I need to put those commands into a user-friendly format for PiKISS users. I hope to accomplish this soon.

Thanks again!

jmcerrejon, 2024-12-25:
Added f0e37f52f03584c1006f2ac72f07a77397f17402

closed 2024-12-25 by jmcerrejon