jmcerrejon/PiKISS #231: several games/apps wont work

qwert2763, 2025-02-15:
srb2, vvvvvv just dont boot, and just immediately closes the terminal when i open the app through there. they both have the required data files but just dont work snes9x doesnt seem to work either, but it does boot. is there supposed to be a gui because i dont have one. if its supposed to not have a gui, how do i load roms? ps im on a pi 4 4gb ram pi os fully updated

jmcerrejon, 2025-02-16:
snes9x doesn't have any GUI. Maybe it exists some project to give it to the emulator, but I don't know any of them.
I'll check out the apps that are not running. Thanks for the report.

jmcerrejon, 2025-11-22:
I'm gonna close this issue. I'm updating game sections for aarch64 compatibility. Later, I'll keep the work up following emulators,... Open a new issue for specific game/emu when It's done. 

Thank you! 😄

closed 2025-11-22 by jmcerrejon