jmcerrejon/PiKISS #234: 2 ship 2 harkinian port (The Legend of Zelda Majora's Mask N64 )

AndresJosueToledoCalderon, 2025-05-29:
So I was wondering if you may want to add this game to pikiss. It is already compiled in the .zip archive. It just needs a few dependecies / libraries to run.

It is compiled on raspberry pi os 64 bit.

Actually the only problem is that the assets need to be provided by the owner (basically the rom).

https://github.com/AndresJosueToledoCalderon/Compile-2Ship2Harkinian-for-Raspberry-Pi

[https://github.com/AndresJosueToledoCalderon/Compile-2Ship2Harkinian-for-Raspberry-Pi](2 ship 2 harkinian)

AndresJosueToledoCalderon, 2025-05-29:
ok, I forgot how to use the url tool, but both are the same...

jmcerrejon, 2025-06-01:
Thanks, man! It was so easy. 😍
Join the Discord and talk with me.
Add at 47bc927ecbfe86e54658c08bad0e3e78613a49cc

closed 2025-06-01 by jmcerrejon