jmcerrejon/PiKISS #235: Diablo1 Failed installation

Pangel71, 2025-07-10:
Diablo 1
========
· The game needs the original game files to play it, but a shareware version will be installed.
· Thanks to DevilutionX for the port.

Installing, please wait...

Downloading...

devilutionx-linux-a 100%[===================>]  10.39M  8.66MB/s    in 1.2s    

Installing necessary custom package...
dpkg-deb: error: '/tmp/devilutionx-linux-armhf.zip' is not a Debian format archive
dpkg: error processing archive /tmp/devilutionx-linux-armhf.zip (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /tmp/devilutionx-linux-armhf.zip

jmcerrejon, 2025-07-10:
Mmm, It's work on my side: Raspberry Pi OS 64-bit.
Are you using the 32-bit version? Maybe it's the case.

Pangel71, 2025-07-10:
It seems to me like devilutionx does not have the filename: "devilutionx-linux-armhf.zip" in version 1.54 (the latest version).
But I am using a 32-bit and I can try to change to 64-bit if you think that will help.

jmcerrejon, 2025-07-10:
> It seems to me like devilutionx does not have the filename: "devilutionx-linux-armhf.zip" in version 1.54 (the latest version).
> But I am using a 32-bit and I can try to change to 64-bit if you think that will help.

It works on 64-bit. I'll try to fix for 32-bit

Pangel71, 2025-07-15:
I have now installed Raspberry Pi OS 64-bit and Diablo 1 works fine.
I was just wondering if I want to play Diablo II, I would need to change back to Pi OS 32-bit for that to work?
Thanks for your help.